lancaster-university/microbit-docs #16: Are the Bluetooth Services enabled by default?

danjenkins, 2016-04-12:
I can't seem to find out why I'm erroring when trying to instantiate the BLE Button Service

```
error: no matching function for call to 'MicroBitButtonService::MicroBitButtonService()'
```

with this being the call to it.

```
int main()
{
    uBit.init();

    new MicroBitButtonService();

    release_fiber();
}
```

I'm using the DAL as a target and the docs don't seem to be pointing me towards anything. I checked the config for something that I needed to enable but couldn't find anything.

finneyj, 2016-04-12:
hi @danjenkins 

BLE should normally be enabled, but the default security settings are very high (inline with BBC requirements for child protection etc.), so that might be throwing your BLE device a bit.

For testing and development, there's a MICROBIT_BLE_OPEN flag in inc/core/MicroBitConfig.h

Can you try setting that to a '1', then doping a clean rebuild?

You also might need to give a bit more memory to the BLE stack if you're running more services, so maybe increase MICROBIT_SD_GATT_TABLE_SIZE a bit while you're in there - say 0x500.

--joe

danjenkins, 2016-04-12:
Thanks for replying so quickly @finneyj !

So I upped the table size and enabled BLE_OPEN, did a `yt update`, `yt clean` and `yt build` and its still failing...

```
/Users/danjenkins/git/microbit-button-ble/source/Button-BLE.cpp: In function 'int main()':
/Users/danjenkins/git/microbit-button-ble/source/Button-BLE.cpp:11:31: error: no matching function for call to 'MicroBitButtonService::MicroBitButtonService()'
     new MicroBitButtonService();
                               ^
In file included from /Users/danjenkins/git/microbit-button-ble/yotta_modules/microbit-dal/inc/bluetooth/MicroBitBLEManager.h:57:0,
                 from /Users/danjenkins/git/microbit-button-ble/yotta_modules/microbit/inc/MicroBit.h:65,
                 from /Users/danjenkins/git/microbit-button-ble/source/Button-BLE.cpp:1:
/Users/danjenkins/git/microbit-button-ble/yotta_modules/microbit-dal/inc/bluetooth/MicroBitButtonService.h:52:5: note: candidate: MicroBitButtonService::MicroBitButtonService(BLEDevice&)
     MicroBitButtonService(BLEDevice &_ble);
     ^
/Users/danjenkins/git/microbit-button-ble/yotta_modules/microbit-dal/inc/bluetooth/MicroBitButtonService.h:52:5: note:   candidate expects 1 argument, 0 provided
/Users/danjenkins/git/microbit-button-ble/yotta_modules/microbit-dal/inc/bluetooth/MicroBitButtonService.h:43:7: note: candidate: constexpr MicroBitButtonService::MicroBitButtonService(const MicroBitButtonService&)
 class MicroBitButtonService
       ^
/Users/danjenkins/git/microbit-button-ble/yotta_modules/microbit-dal/inc/bluetooth/MicroBitButtonService.h:43:7: note:   candidate expects 1 argument, 0 provided
/Users/danjenkins/git/microbit-button-ble/yotta_modules/microbit-dal/inc/bluetooth/MicroBitButtonService.h:43:7: note: candidate: constexpr MicroBitButtonService::MicroBitButtonService(MicroBitButtonService&&)
/Users/danjenkins/git/microbit-button-ble/yotta_modules/microbit-dal/inc/bluetooth/MicroBitButtonService.h:43:7: note:   candidate expects 1 argument, 0 provided
ninja: build stopped: subcommand failed.
error: command ['ninja'] failed
```

I'm not sure if I'm just not doing something which you'd class a standard... this is my first mbed experience etc.

finneyj, 2016-04-12:
Hi @danjenkins 

apologies - rushing here... you need to supply the BLE stack as a parameter to the constructor. Please see the "Enabling the Service" section here:

http://lancaster-university.github.io/microbit-docs/ble/button-service/

danjenkins, 2016-04-12:
Now I feel stupid, I must have removed that earlier while I was trying to debug something. Thanks so much @finneyj !

finneyj, 2016-04-12:
@danjenkins 

hey, no worries! Let us know if it works - we're still ironing out support on mbed.org, so please let us know of any issues. :-)

closed 2016-05-09 by jamesadevine