lancaster-university/microbit-docs #37: getAnalogValue should be 0- 1023

drmpf, 2016-10-18:
https://lancaster-university.github.io/microbit-docs/ubit/io/#getanalogvalue  shows result as 0 to 1024 but nRF51 table 322 shows result as 0 to 1023

Note  setAnalogValue is also shown in the docs as 0 - 1024, I assume this should also be 0 - 1023

bagofatoms, 2017-01-09:
1023 is also my understanding. Also I think the description is unclear, wrong or misleading. The description talks about analog levels (both read and write), but I'm suspecting its actually a pulse in a period rather than a level.

Darshnik253, 2022-02-02:
i am making robotic car project in that i have to control dc motor with pwm signal using microbit. so i have written code for that which i combined digital write and analog write pins which part i am attaching below. But code is working only one time after that its not working(error - microbit led is flickering). how can i resolve this error?

pins.digital_write_pin(DigitalPin.P1,0)
pins.digital_write_pin(DigitalPin.P2,0)
pins.digital_write_pin(DigitalPin.P8,1)
pins.digital_write_pin(DigitalPin.P12,0)
pins.analog_write_pin(AnalogPin.P0,511)
pins.analog_set_period(AnalogPin.P0,200)
pins.analog_write_pin(AnalogPin.P3,511)
pins.analog_set_period(AnalogPin.P3,200)