lancaster-university/microbit-docs #43: LED Matrix State confusion

kenyoncoder, 2017-03-02:
There is a serious discrepancy between:
[https://lancaster-university.github.io/microbit-docs/ble/profile/#using-the-bluetooth-profile](url)
and
[https://lancaster-university.github.io/microbit-docs/resources/bluetooth/bluetooth_profile.html](url)

when describing how the state of the LED Matrix is represented. The description in the second document seems to represent the reality of the situation.

finneyj, 2017-03-02:
Thanks both!

bluetooth-mdw, 2017-03-03:
Hopefully this addresses this issue:

https://github.com/lancaster-university/microbit-docs/pull/44

finneyj, 2017-03-03:
Thanks @bluetooth-mdw. Looks good to me.

@kenyoncoder - Do you think this addresses all the errors / ambiguities you saw?

kenyoncoder, 2017-03-03:
Thanks @bluetooth-mdw  and @finneyj  - looks excellent to me.

finneyj, 2017-03-03:
great - thanks guys. Just merged that change so closing off this issue.

closed 2017-03-03 by finneyj

kenyoncoder, 2017-03-04:
@finneyj 

Silly question - when will the change appear live on the Lancaster website?

jamesadevine, 2017-03-04:
@kenyoncoder should be there now.

kenyoncoder, 2017-03-04:
oops - sorry, now it's live, there's still a bit of the out of date description present:

> **Characteristics**
LED Matrix State : Allows the state of any|all LEDs in the 5x5 grid to be set to on or off with a single GATT operation. Consists of a 32 bit field with bits 0 - 24 representing the off (0) or on (1) state of the corresponding LED.

Wasn't obvious when I checked the change done by @bluetooth-mdw