lancaster-university/microbit-docs #55: Mbed online compilation

microbit-mark, 2018-10-05:
From support: https://support.microbit.org/a/tickets/17900

Are these instructions still correct?

https://lancaster-university.github.io/microbit-docs/online-toolchains/#hello-world-on-mbedorg

By following them I get:

Error: No linker script found

[image]

Should micro:bit scripts still compile here. Do we have a working example?

microbit-mark, 2018-10-08:
It looks like it may be broken https://stackoverflow.com/a/52598213

microbit-carlos, 2018-10-08:
The online mbed compiler is currently not working for Mbed 2 targets like the micro:bit.
The given instructions to use the offline mbed cli is not working either: https://github.com/ARMmbed/mbed-cli/issues/327

Users should be directed to use Yotta instead as per: https://lancaster-university.github.io/microbit-docs/offline-toolchains/