lancaster-university/microbit-docs #68: Correction in BLE profile page

microbit-carlos, 2018-12-05:
In this page https://lancaster-university.github.io/microbit-docs/ble/profile/#gatt-services it currently reads:

> Known issue: the firmware on micro:bits shipped initially to schools contains the full Bluetooth profile. Once the software has been changed by flashing an application created using any of the on-line code editors however, all services are lost except for Device Information Service, Event Service and DFU Control Service. 

Only 4 services are actually enabled in the OOB experience hex:
https://github.com/lancaster-university/microbit-samples/blob/c3b02564fd8082f3bf80472e679aa2f9b707c7c2/source/examples/out-of-box-experience/main.cpp#L630-L633

And applications created with the Online Python editor will not have any BLE services at all.

So we need to rephrase a bit.

microbit-carlos, 2018-12-05:
Also, there is this note which is not clear what service it is referring to:

> Note: This service was added after micro:bits initially shipped to school and so is not in the default image. |

microbit-sam, 2018-12-10:
Would it make more sense to remove the text referring to the initial school drop? It's been 3(?) years
The rest can be changed to refer to the current OOB

RE: Python. Would a sentence in the introduction explaining that BLE is unavailable when using Python be enough?

carlosperate, 2019-01-29:
@microbit-sam was this updated? Can it be closed?

microbit-sam, 2019-01-29:
Doesn't seem to be live, I think the work is done on a branch that I'm trying to track down