lancaster-university/microbit-docs #76: New BLE characteristic for mag calibration needs documenting

carlosperate, 2019-02-22:
More info in https://github.com/lancaster-university/microbit-dal/issues/412#issuecomment-448186568.

@microbit-sam is there anything specific about this characteristic that needs to be included as well?

carlosperate, 2019-02-22:
Btw, @microbit-sam is the change to the original magnetometer service not longer triggering a calibration automcatically documented somewhere?

microbit-sam, 2019-02-25:
It looks like this was originally submitted as a change to the docs in this PR https://github.com/lancaster-university/microbit-docs/pull/48 but never merged

I'm not sure there's a reference to automatic calibration no longer happening, will check through and update that existing PR

microbit-carlos, 2019-02-25:
Oh right, I didn't see #48.
Not sure you'll be able to update that PR with more info about the changhe, but I think the changelog included in #48 is probably enough.

ukBaz, 2019-02-25:
The characteristic is also missing from:
https://lancaster-university.github.io/microbit-docs/resources/bluetooth/bluetooth_profile.html

microbit-sam, 2019-02-26:
#48 has been merged and published
https://lancaster-university.github.io/microbit-docs/resources/bluetooth/bluetooth_profile.html

I'll figure out where best to document that calibration no longer happens on start up, I don't think it's documented that it did, but seeing as the functionality has changed it ought to be there