lancaster-university/microbit-docs #80: Outdated micro:bit Bluetooth profile

ngammarano, 2020-10-14:
The micro:bit Bluetooth profile defined [here](https://lancaster-university.github.io/microbit-docs/resources/bluetooth/bluetooth_profile.html) is outdated.

- the Partial Flashing service is not included
- the Manufacturer Number String characteristic appears as Mandatory
- the UART service appears as Mandatory
- the Event service appears as Optional, while a Bluetooth micro:bit program made in MakeCode always includes it

Since many projects refer to that link to base on the micro:bit Bluetooth profile, I think it should be updated.

bluetooth-mdw, 2020-10-14:
@ngammarano I agree. I think I've flagged this before.

the Partial Flashing service is not included
Agree 

the Manufacturer Number String characteristic appears as Mandatory
Agree (this looks like an error in the documentation)

the UART service appears as Mandatory
Agree (this also looks like an error)

the Event service appears as Optional, while a Bluetooth micro:bit program made in MakeCode always includes it
Disagree - it's optional but MakeCode chooses to always include it. MakeCode is not the only user of the profile and personally I wish MakeCode allowed you to exclude this service and others it insists on including.

ngammarano, 2020-10-28:
The micro:bit Bluetooth profile defined [here](https://lancaster-university.github.io/microbit-docs/resources/bluetooth/bluetooth_profile.html) says that the valid values for the magnetometer period are:
> `Valid values are 1, 2, 5, 10, 20, 80, 160 and 640.`

However, in practice, the valid values I get are 12, 25, 50, 100, 200, 400, 800, 1600, 3200, 6400 and 12800, which are consistent with those defined in [the magnetometer MAG3110 C++ driver file](https://github.com/lancaster-university/microbit-dal/blob/master/source/drivers/MAG3110.cpp#L40) and in [the magnetometer MAG3110 datasheet](https://www.nxp.com/docs/en/data-sheet/MAG3110.pdf) (see Table 32).

bluetooth-mdw, 2020-10-28:
Those were that values agreed with Lancaster University at the time. They (or whoever maintains the DAL) need to decide whether the profile's implementation should be validating values provided in characteristic writes and rejecting values not in the agrees range or whether the documentation needs updating (again, by whoever it is that should be maintaining it - FYI, it's not me).

ngammarano, 2020-11-06:
> Those were that values agreed with Lancaster University at the time. They (or whoever maintains the DAL) need to decide whether the profile's implementation should be validating values provided in characteristic writes and rejecting values not in the agreed range or whether the documentation needs updating (again, by whoever it is that should be maintaining it - FYI, it's not me).

OK, then where should I report this?

ngammarano, 2020-11-06:
The micro:bit Bluetooth profile defined [here](https://lancaster-university.github.io/microbit-docs/resources/bluetooth/bluetooth_profile.html) says that the Pin AD Configuration and the Pin IO Configuration characteristics of the IO Pin Service have 1 field of type `uint24` named Pin_AD_Config_Value and Pin_IO_Config_Value respectively.

However, when I read those characteristics, I get `uint32`, which is consistent with [the IO Pin characteristics in the DAL](https://github.com/lancaster-university/microbit-dal/blob/master/source/bluetooth/MicroBitIOPinService.cpp#L169).

Is this an error of the Bluetooth profile?